drunken-octopus/drunken-octopus-marlin #1: Delete old releases

marciot, 2019-10-20:
There are a bunch of old releases in the releases page. These are from Marlin upstream and need to be deleted as to not make things overly confusing for users.

crpalmer, 2019-10-20:
I assume that you are not planning on porting any of the command line tools that Lulzbot uses for building?

marciot, 2019-10-20:
@crpalmer: Actually, I do plan on bringing them over. I would like to run them under Windows Subsystem for Linux on my Windows 10 PC.

crpalmer, 2019-10-20:
Oh, that's great.  I really like being able to build the firmware from my linux server via ssh.

marciot, 2019-10-21:
@crpalmer: I am compiling using the build tools right now. So far so good. If you want to start messing around with the code, feel free. The build stuff has been pushed as of a few minutes ago.

marciot, 2019-10-21:
Closing this issue since the first release will be made shortly.

closed 2019-10-21 by marciot