drunken-octopus/drunken-octopus-marlin #2: Feedback on 2.0.0.1-rc1

marciot, 2019-10-21:
If you've given `2.0.0.1-rc1` a try, please post your reactions here. Thank you!

The next few days I'll need to focus on my job search, so I may not be able to respond immediately, but hopefully I will return next week.

crpalmer, 2019-10-22:
I don't have time to test until the end of the month, but you mention that it is "missing the washers".  Is it missing in x/y or missing in z?  A fair bit ago, I tried compiling the lulzbot devel branch for my taz 6 and it missed the washers in z (it probed too high).  I didn't bother debugging it and treated that as a sign the devel branch wasn't ready for my taz 6...

marciot, 2019-10-28:
@crpalmer: I believe the issue with missing in Z is something I actually fixed the last day I was at LulzBot. *Drunken Octopus* should not have that particular problem. The problem I was seeing was that my printer was missing the washers in Y. I replaced the bare steel bed platform on my 1st gen Mini with the nicer looking powder coated one from later generations Mini, but I did not reprint all the other bed components with later versions.  I believe this introduced a coordinate shift in Y. So my Mini is no longer representative of any LulzBot printers. Makes testing firmware a bit difficult.

TechsavvyCNC, 2019-10-29:
It's a shame what lulzbot did to you guys.  I own a taz 6 and a taz pro that I built myself.  Im almost finished building the taz pro.  My taz 6 was having issues on newer firmware where when it went to home Z it would just raise instead of lower into the switch.  I'll test this firmware out on it and let you know how it goes.  The work you've done on marlin is amazing.

marciot, 2019-10-29:
@TechsavvyCNC: I'm pretty sure the issue you saw was the same @crpalmer described and has already been fixed. Please let me know if you find out otherwise!

TechsavvyCNC, 2019-10-29:
Alright.  I'll give it a try tonight.  Thanks

crpalmer, 2019-10-29:
@TechsavvyCNC if it prints once and then fails to level after that because it never touched the probe, you may need to update your start.gcode.  I had problems with that during some firmware upgrade (I forget which version though).

TechsavvyCNC, 2019-10-30:
Flashed firmware to my Taz 6.  When I printer reset it now says PRINTER HALTED RESET.  Even if I try power cycling it does the same thing.  I can get a log for you in a little bit.

marciot, 2019-11-01:
@TechsavvyCNC: Can you get me the serial log?

marciot, 2019-11-16:
@TechsavvyCNC: This should be fixed in the latest release. If it is, please close this one down.

closed 2020-01-27 by marciot