drunken-octopus/drunken-octopus-marlin #4: Misaligned Bedleveling Taz 6 Aerostruder RC2

usulrasolas, 2019-11-04:
### Description

When attempting to run bedleveling, it misses the first by around 4mm in both the x and y postion, with around a 5.5mm offset if measured diagonally from location of nozzle touchdown on buildplate, and the intended location on leveling washer.

### Steps to Reproduce

1. Preheat bed/nozzle
2. Initiate Bedleveling via LCD Menu

**Expected behavior:** [What you expect to happen]
1. Nozzle should touch near "inner edge" of leveling washer and continue onto other washers.

**Actual behavior:** [What actually happens]
1. Nozzle misses by large margin, triggering heat and re-wipe for failed probe

#### Additional Information
Unmodified Taz 6 Using Aerostruder on Drunkenoctopus rc2
[image]

usulrasolas, 2019-11-05:
https://github.com/MarlinFirmware/Marlin/issues/15369

marciot, 2019-11-05:
Yes, it appears as if this is caused by Marlin no longer allowing probe points outside of the bed area. This is not good.

TechsavvyCNC, 2019-11-05:
Yes I can confirm the same issue on my Taz 6 too.

usulrasolas, 2019-11-05:
Having looked around the only work around is to expand the bed size but that could cause other problems with collision. I'll keep improving my understand of the underlying code.

TechsavvyCNC, 2019-11-05:
That's what I was thinking was expanding the bed size but I'm sure theres a better way of doing it.  I'm looking in to it.

ZS1FJK, 2019-11-06:
I Almost got it to work by increasing the bed size from 280 to 300 but ran into some other complications and i really don't have time right now to fix them. I'm sure if i had a day or two i could get it working. 

If you try this you will have to adjust. Z safe zero and Wiper pad .

ZS1FJK, 2019-11-06:
Also your limit switch positions

TechsavvyCNC, 2019-11-11:
Nice.  Will test and see if issue is fixed.

usulrasolas, 2019-11-12:
I can run the manual bed levelling and it completes, the left side is in slightly different place than stock firmware but seems well within acceptable areas. Will have to get ready to calibrate z offset and try a print!

closed 2019-11-12 by usulrasolas