drunken-octopus/drunken-octopus-marlin #10: Error: EEPROM datasize error.

keawepoo, 2020-01-15:
### Description

Send: M501
Recv: Error:EEPROM datasize error.
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: EEPROM datasize error.
Recv: echo:Index: 688 Size: 692
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:Settings Stored (692 bytes; crc 64672)
Recv: echo:EEPROM Initialized

### Steps to Reproduce

Flashed
pluged usb into pi
started octoprint

**Expected behavior:** [What you expect to happen]
to work

**Actual behavior:** [What actually happens]
connects > error > disconnects.

#### Additional Information
patreon build -  Oliveoil_TAZ6 Yellowfin_DualExtruderV3 - stock board

marciot, 2020-01-15:
At the moment, I regret I am only am able to maintain the build system and LulzBot specific modification. This appears to be an error of a more general nature and the best place to report this would be in upstream Marlin. 

In any event, it seems a bit peculiar that OctoPrint is dropping the connection when this happens. An EEPROM error is **not** a critical error. Generally the printer will only load default settings in this case. So perhaps this is something that also should be reported in OctoPrint.

keawepoo, 2020-01-15:
Fixed... 

Refashed with lulzbot cura. 
Originally flashed with Cura 4.1

closed 2020-01-15 by keawepoo