drunken-octopus/drunken-octopus-marlin #11: Feature Request: Separate x and y steps from bed leveling corner locations

jim2386, 2020-03-05:
Hi Marcio,

I hope this is the right place to put this. One of my biggest pet peeves with Lulzbot firmware is that the g29 probe locations are hard coded into the firmware. Sometimes the nozzle doesn't probe the corner of the washer as it should. The main time this happens is when I change x and y steps on the LCD. I use X and Y steps to calibrate cubes to ensure my parts are dimension-ally accurate.

However, when I change the X and Y steps, this inadvertently moves the G29 washer probe location because the firmware (I assume) goes over so many steps and assumes the washer to be there. If there was a way to either do the following:

1. Be able to set the X and Y steps for printed parts but the number of steps required to reach the bed corners stays the same as originally spec'd by lulzbot.

or (and better)

2. Be able to set X and Y steps for printed parts AND be able to separately set the number of steps required to get to the washers. Instead of R155, we could set it to R157 or B0 instead of B2....R = Right side, B = back washers. The only way I know how to do this today would be to go into the firmware, find the probe locations, change them, recompile and make my own firmware.

I've seen enough people struggle where the nozzle doesn't quite hit the top of the washer, or the nozzle moves too far and hits the screw head in the middle of the washer.

Is there a way to add a feature to set probe locations without a firmware recompile by the user?

Thanks!
Jim

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

marciot, 2020-03-16:
A lot of things in Marlin are hard coded because the 8-bit micros simply do not have enough RAM for run-time configuration of very many parameters.

Furthermore, adjusting probe points on the fly does not make sense. What Marlin records is the physical position of the probe points, in millimeters, not the steps between probe points. This does not change and can, under ordinary circumstances, be compiled into the firmware.

My guess is that you have something older than a Mini 2 or a TAZ Workhorse/Pro? For older printers, when LulzBot made their firmware, LulzBot incorrectly fudged the steps in order to account for ABS shrinkage. They set the steps to 100.5 when in reality it ought to be 100. This is absolutely incorrect and as a consequence the default steps per mm are incorrect, which also means that the recorded distance between probe points is incorrect to compensate.

The correct way to solve this would be to measure the actual motion of the toolhead and set the steps so that the toolhead is moving the correct amount in millimeters. This should be measured at the toolhead, not by measuring a calibration print, as compensating for material characteristics is something that should be done in the slicer, not the firmware. Once the printer is calibrated correctly, then one could measure the actual distance between probe points on the bed and hard code that into the firmware. Everything would work at this point.

However, given that LulzBot did it incorrectly in the first place, you run into the problem you are describing if you try to change the steps. If you give me the *correct* number of steps for the printer, I may be able to adjust the probe points in drunken-octopus.

marciot, 2020-05-07:
The latest firmware adjust the X and Y steps so it is closer to true and gives more dimensionally accurate parts. That firmware also adjusts the probe points. This should allow you to reach all probe points without problems. Please give it a try and let me know whether it works better.

closed 2020-09-15 by marciot