drunken-octopus/drunken-octopus-marlin #13: April Release - Taz6 v3 BLTouch - Fails to upload(?) - Patreon Edition

keawepoo, 2020-04-06:
### Steps to Reproduce
1. Start Cura
2. Update firmware

**Expected behavior:** 
To update the firmware...

**Actual behavior:** 
It just sits at "updating firmware"... For at least 5mins. After that I canceled the update.

#### Additional Information
https://imgur.com/a/xvHd3RX

marciot, 2020-04-08:
The recommended flashing program for the Archim is BOSSA (https://www.shumatech.com/web/products/bossa).

With it, follow this procedure for flashing:

- Open up the Windows device manager and open "Port (COM & LPT)".
- Press and hold the "Erase" button while you tap the "Reset" button. The Device Manager should refresh show a "Bossa Program Port".
- Upload FW to that port using BOSSA with "Erase all" and "Boot to flash" checked.
- Tap the Reset button after the flash completes.
- The Device Manager should refresh again and it will show "Serial GCode 3D Printer" on a different port.

If you cannot connect to the serial port after this process, try the following:

- Try manually connecting to that port at 250000 baud using PuTTY for Windows.
- Uninstalled and deleted the drivers that windows allocates to the serial port.
- Repeat steps above.

keawepoo, 2020-04-08:
Im sorry, I thought I had listed it as the stock board. Will this still work?

marciot, 2020-04-08:
Oh, I'm sorry. No, those procedures are for the Archim board. For a stock (Rambo) board, you could try either one of these tools:

- https://www.hobbytronics.co.uk/arduino-xloader
- https://blog.zakkemble.net/avrdudess-a-gui-for-avrdude/

I've used the first tool, but not the second. Anyhow, the fact Cura isn't able to load the FW isn't really anything to do with Drunken Octopus. Have you tried asking for help in the Cura forums?

keawepoo, 2020-04-08:
Its all good, It was my own oversight. What board is it listed as? Mega or Uno? Im using the xloader.
Well I also tryed lulzbot one too and it failed. That being somewhat old I dont know if it would still be a Cura issue.

marciot, 2020-04-08:
Mega.

keawepoo, 2020-04-08:
Baub rate 57600 or 250000??

marciot, 2020-04-08:
I think either will work, but 250000 preferred.

keawepoo, 2020-04-08:
at 250000, it says that upload has failed... at 57600 it just stays at "uploading..."

marciot, 2020-04-08:
Does the printer otherwise turn on? You may have blown fuses on the board or something. Make sure the BLTouch didn't short something out. I recommend disconnecting it, as well as anything else you might have connected to the board.

marciot, 2020-04-08:
You could also reach out to Ultimachine for ideas. At this point, it doesn't seem like it's a firmware issue, but a hardware issue of some sort.

keawepoo, 2020-04-08:
BLTouch isnt even connected yet... Ive tried to print via octoprint and it works. but ya will do. 
Thank you.

marciot, 2020-04-08:
That's weird. Usually flashing is a fairly straightforward process, easier on the Rambo than the Archim. Not sure why your board is being problematic.

keawepoo, 2020-04-08:
Well I do know that theres a EPROM issue... where it doesnt save anything. I think I posted awile back about it.

keawepoo, 2020-04-08:
> That's weird. Usually flashing is a fairly straightforward process, easier on the Rambo than the Archim. Not sure why your board is being problematic.

So I just flashed lulzbot firmware on and it worked... So that might be a issue with DO?

marciot, 2020-04-08:
Weird. Can you try the regular TAZ 6 build from DO? 

- Marcio

keawepoo, 2020-04-08:
I just did. I tryed the R10 standard Tilapia_SingleExtruder, That one failed. Then I tryed R9 standard Tilapia_SingleExtruder and it worked...

So it really looks like theres a issue over all in R10.

marciot, 2020-04-08:
Interesting. I’ll see if I can look into it this weekend. Thanks.

keawepoo, 2020-04-12:
Fixed with R11.

closed 2020-04-12 by keawepoo