drunken-octopus/drunken-octopus-marlin #14: April Release R11 - Taz6 Rambo v3 BLTouch - Extruders Mixed up - Patreon Edition

keawepoo, 2020-04-12:
### Description
Extruders are mixed up. So T0 is T1, and T1 is T0.

### Steps to Reproduce
1. Update to R11...
2. Heat Extruder 1 up.
3. Place PLA on Extruder 1, Then Extruder 2.

**Expected behavior:** [What you expect to happen]
Extruder 1 to heat up and not Extruder 2....

**Actual behavior:** [What actually happens]
Extruder 2 heats up....

#### Additional Information
Continuation to https://github.com/marciot/drunken-octopus-marlin/issues/13 ?
https://drive.google.com/open?id=13tf3qWaKU7p8icwCepTZw8bgXK0UvoSj - R11
https://drive.google.com/open?id=1DqEC7T1To03Jw52oyeWjsw0TgzjVZNRa - Lulzbot Firmware
https://drive.google.com/open?id=1hHL7qnHIqo7WpRjagSLiOY62gkc-Cfuk - Wiring

Thank you.

marciot, 2020-04-12:
This is not a bug, at least not a bug in the firmware. LulzBot wired up the v3 with the two extruders backwards and relied on a firmware hack to cover for the mistake. This was an ugly workaround and a far better solution is to simply cross the wires when plugging in the toolhead. So in drunken-octopus, T0 is the same extruder that is used for the single toolheads. This means that for the v3, you need to cross the wires if you want it to match up with the labels on the toolhead itself.

keawepoo, 2020-04-12:
By chance you know what wire I have to cross?

marciot, 2020-04-12:
@keawepoo: Yeah, when you plug in the toolhead just swap the two molex connectors the go into the toolhead.

closed 2020-04-13 by keawepoo