drunken-octopus/drunken-octopus-marlin #15: M48 not enabled for z probe repeatability test in binaries

hrabbot, 2020-04-16:
It appears that M48 command is not recognized in the current binaries for Taz 5 running on Rambo and Archim2 with BLTouch.  Z_MIN_PROBE_REPEATABILITY_TEST may need to be enabled.

marciot, 2020-04-27:
Okay. I can add this in the next release!

hrabbot, 2020-05-03:
> 
> 
> Okay. I can add this in the next release!

It would also perhaps be a good idea to slow the z speed a bit (maybe halved?) during probing, as I'm getting kinda sketchy mesh results with genuine BLTouch and offbrand 3DTouch.

marciot, 2020-05-07:
The probe speed has been extensively tested. Slower or faster speeds gave worse results in our testing.
Z_MIN_PROBE_REPEATABILITY_TEST has been enabled in the latest release.

closed 2020-05-07 by marciot