drunken-octopus/drunken-octopus-marlin #21: Enabling USE_CONTROLLER_FAN breaks compilation in Arduino IDE

SwiftNinja, 2020-09-03:
As the title states. It's a problem in DrunkenOctopus, and, it appears, Marlin Upstream. 

                 #define USE_CONTROLLER_FAN

... breaks the compilation. The compiler will not allow a constant expression to be assigned (what it perceives as) a dynamic value. The line of code that is the issue is in src/feature/controllerfan.h:

                  static const controllerFan_settings_t constexpr &settings = controllerFan_defaults; 

There is a workaround, but the workaround shouldn't be necessary. The workaround is, if you enable USE_CONTROLLER_FAN, enable:
                 
                   #define CONTROLLER_FAN_EDITABLE 

and then disable the menu item

                    //#define CONTROLLER_FAN_MENU          

... and it will now compile, but technically M70 is still enabled through this, which isn't the desired behavior.

Steps to reproduce:

1. #define USE_CONTROLLER_FAN // in config.h
2. Compile in Arduino IDE
3. Result:

In file included from sketch/src/MarlinCore.cpp:205:0:
sketch/src/feature/controllerfan.h:61:67: error: non-constant in-class initialization invalid for static member 'ControllerFan::settings'
       static const controllerFan_settings_t constexpr &settings = controllerFan_defaults;
                                                                   ^
sketch/src/feature/controllerfan.h:61:67: error: (an out of class initialization is required)
**sketch/src/feature/controllerfan.h:61:67: error: 'ControllerFan::settings' cannot be initialized by a non-constant expression when being declared**
exit status 1
Error compiling for board Archim.

marciot, 2020-09-15:
@SwiftNinja: I am able to reproduce this error on Arduino IDE 1.9.0 beta, which uses GCC 4.8.3, but the error is not present in avr-gcc 6.3.1, which is what I use to compile drunken-octopus. Overall, the code looks correct to me, so this is more a matter of the Arduino IDE using an ancient compiler.

The code could be made less correct in order to make it compatible with the Arduino IDE, but if that is the case, it's a decision to be made by Marlin developers. Since they use platform.io for compilation, I don't think they would be willing to make a change, but you can try making an issue upstream and see if it gets any traction. Another place to report this would be Arduino, to see if they can update their compilers.

I am closing this issue here since it's does not really a Drunken Octopus issue.

closed 2020-09-15 by marciot

SwiftNinja, 2020-09-15:
Good grief. I figured it as a GCC versioning issue, since that failing syntax allowance did seem to have been a change spec for GCC. But I wasn't aware that Arduino IDE was referencing an _ancient_ version of GCC; I figured it was automatically using whatever release of GCC was installed by default.

I couldn't understand how this "bug" had persisted so long in Marlin upstream - I discovered it months ago but figured someone would find it quickly without opening a case. Now at least I understand what the core disconnect is, I'll see if there's a workaround for Arduino.

Thanks again for your investigation and thoughtful response. Cheers.