drunken-octopus/drunken-octopus-marlin #26: RC22 - G26 is called on stock startup code but function is different

IggyTheWolf, 2020-09-13:
### Description

The stock start gcode for the Taz 6 includes the G26 to "clear potential 'probe fail' condition". It appears Marlin now uses it to validate the bed leveling mesh. 

### Steps to Reproduce

This is on a Taz6: Marlin_Oliveoil_TAZ6ArchimBLTouch_Angelfish_Aerostruder_2.0.0.1_9a338bc1c.bin

**Expected behavior:**

Here is the factory startup code for the Taz6

;This G-Code has been generated specifically for the LulzBot TAZ 6 with Aerosturder
M73 P0 ; clear GLCD progress bar
M75 ; start GLCD timer
G26 ; clear potential 'probe fail' condition
M107 ; disable fans
M420 S0 ; disable leveling matrix
G90 ; absolute positioning
M82 ; set extruder to absolute mode
G92 E0 ; set extruder position to 0
M140 S{material_bed_temperature_layer_0} ; start bed heating up
G28 XY ; home X and Y
G1 X-19 Y258 F1000 ; move to safe homing position
M109 R{material_soften_temperature} ; soften filament before homing Z
G28 Z ; home Z
G1 E-15 F100 ; retract filament
M109 R{material_wipe_temperature} ; wait for extruder to reach wiping temp
G1 X-15 Y100 F3000 ; move above wiper pad
G1 Z1 ; push nozzle into wiper
G1 X-17 Y95 F1000 ; slow wipe
G1 X-17 Y90 F1000 ; slow wipe
G1 X-17 Y85 F1000 ; slow wipe
G1 X-15 Y90 F1000 ; slow wipe
G1 X-17 Y80 F1000 ; slow wipe
G1 X-15 Y95 F1000 ; slow wipe
G1 X-17 Y75 F2000 ; fast wipe
G1 X-15 Y65 F2000 ; fast wipe
G1 X-17 Y70 F2000 ; fast wipe
G1 X-15 Y60 F2000 ; fast wipe
G1 X-17 Y55 F2000 ; fast wipe
G1 X-15 Y50 F2000 ; fast wipe
G1 X-17 Y40 F2000 ; fast wipe
G1 X-15 Y45 F2000 ; fast wipe
G1 X-17 Y35 F2000 ; fast wipe
G1 X-15 Y40 F2000 ; fast wipe
G1 X-17 Y70 F2000 ; fast wipe
G1 X-15 Y30 Z2 F2000 ; fast wipe
G1 X-17 Y35 F2000 ; fast wipe
G1 X-15 Y25 F2000 ; fast wipe
G1 X-17 Y30 F2000 ; fast wipe
G1 X-15 Y25 Z1.5 F1000 ; slow wipe
G1 X-17 Y23 F1000 ; slow wipe
G1 Z10 ; raise extruder
M109 R{material_probe_temperature} ; wait for extruder to reach probe temp
G1 X-9 Y-9 ; move above first probe point
M204 S100 ; set probing acceleration
G29          ; start auto-leveling sequence
M420 S1      ; enable leveling matrix
M425 Z       ; use measured Z backlash for compensation
M425 Z F0    ; turn off measured Z backlash compensation. (if activated in the quality settings, this command will automatically be ignored)
M204 S500 ; restore standard acceleration
G1 X0 Y0 Z15 F5000 ; move up off last probe point
G4 S1 ; pause
M400 ; wait for moves to finish
M117 Heating... ; progress indicator message on LCD
M109 R{material_print_temperature_layer_0}  ; wait for extruder to reach printing temp
M190 R{material_bed_temperature_layer_0} ; wait for bed to reach printing temp
G1 Z2 E0 F75 ; prime tiny bit of filament into the nozzle
M117 TAZ 6 Printing... ; progress indicator message on LCD

Here is the startup code I am using slightly modified for the new mesh bed leveling and BLTouch. Note I commented out G26 after discovering this error.

;Iggy Project - thing:4498052
;This G-Code has been generated specifically for the LulzBot TAZ 6 with Aerosturder, Drunken Octopus firmware, and BLTouch
;This G-Code must be run after a new UBL mesh is created and stored.
;Run UBL_Taz6_DO.gcode first
M73 P0 ; clear GLCD progress bar
M75 ; start GLCD timer
;G26 ; clear potential 'probe fail' condition
M107 ; disable fans
M420 S0 ; disable leveling matrix
G90 ; absolute positioning
M82 ; set extruder to absolute mode
G92 E0 ; set extruder position to 0
M140 S{material_bed_temperature_layer_0} ; start bed heating up
G28 XY ; home X and Y
G1 X-19 Y258 F1000 ; move to safe homing position
M109 R{material_soften_temperature} ; soften filament before homing Z
G28 Z ; home Z
G1 E-15 F100 ; retract filament
M109 R{material_wipe_temperature} ; wait for extruder to reach wiping temp
G1 X-15 Y100 F3000 ; move above wiper pad
G1 Z1 ; push nozzle into wiper
G1 X-17 Y95 F1000 ; slow wipe
G1 X-17 Y90 F1000 ; slow wipe
G1 X-17 Y85 F1000 ; slow wipe
G1 X-15 Y90 F1000 ; slow wipe
G1 X-17 Y80 F1000 ; slow wipe
G1 X-15 Y95 F1000 ; slow wipe
G1 X-17 Y75 F2000 ; fast wipe
G1 X-15 Y65 F2000 ; fast wipe
G1 X-17 Y70 F2000 ; fast wipe
G1 X-15 Y60 F2000 ; fast wipe
G1 X-17 Y55 F2000 ; fast wipe
G1 X-15 Y50 F2000 ; fast wipe
G1 X-17 Y40 F2000 ; fast wipe
G1 X-15 Y45 F2000 ; fast wipe
G1 X-17 Y35 F2000 ; fast wipe
G1 X-15 Y40 F2000 ; fast wipe
G1 X-17 Y70 F2000 ; fast wipe
G1 X-15 Y30 Z2 F2000 ; fast wipe
G1 X-17 Y35 F2000 ; fast wipe
G1 X-15 Y25 F2000 ; fast wipe
G1 X-17 Y30 F2000 ; fast wipe
G1 X-15 Y25 Z1.5 F1000 ; slow wipe
G1 X-17 Y23 F1000 ; slow wipe
G1 Z10 ; raise extruder

;M204 S500 ; set probing acceleration

G29 A	;Activate the UBL System.
G29 L1       ;Load a mesh from EEPROM slot1
G29 J2       ;Probe 4 points and tilt the mesh according to what it finds.

M425 Z       ; use measured Z backlash for compensation
M425 Z F0    ; turn off measured Z backlash compensation. (if activated in the quality settings, this command will automatically be ignored)

;M204 S500 ; restore standard acceleration

G1 X0 Y0 Z15 F5000 ; move up off last probe point
G4 S1 ; pause
M400 ; wait for moves to finish
M117 Getting HOT HOT HOT... ; progress indicator message on LCD
M109 R{material_print_temperature_layer_0}  ; wait for extruder to reach printing temp
M190 R{material_bed_temperature_layer_0} ; wait for bed to reach printing temp
G1 Z2 E0 F75 ; prime tiny bit of filament into the nozzle
M121 ; Turn off limit swiches to avoid shifting
M117 TAZ 6 Printing... ; progress indicator message on LCD

marciot, 2020-09-13:
@IggyTheWolf: This is only part of the problem, no? You reported a thermal runaway condition as well?

marciot, 2020-09-13:
@IggyTheWolf: If it were just the G26, I may be inclined to let people simply change their start GCODE. But if something else is going on...

IggyTheWolf, 2020-09-13:
That is correct. I don't put the thermal runaway in here because I've been trying to see if it will do it again, and if so, see if it's related to the g command. It's more serious so I want to make sure I can duplicate it.

How it happened originally is I was trying to print a part with G26 uncommented in the code above. After several restarts the Taz would hang on the Marlin boot screen and once I started one print it hung on heating the extruder. I could tell it was getting too hot because the PLA was oozing and I turned it off. On startup I got the thermal runaway alert.

marciot, 2020-09-13:
@IggyTheWolf: Okay. Let me know what you find out. I am rolling in a fix for the G26 issue

IggyTheWolf, 2020-09-14:
I've been trying to replicate thermal runaway issue and I have not had any success. I guess that's a good thing? I have been been able to replicate the G26 error with mixed results. This is with a 20 min print with G26 in the code.

- Load RC22
- Set probe offset and E steps
- 1st print - successful
- 2nd print - G26 mesh validation ran
- Machine Off
- Machine On
- 3rd print - pad wipe failed and wiped off pad over washer
- Machine off
- Machine on
- 4th print - homing failed. homed y max instead of y min
- machine off
- machine on
- 5th print - homing failed, bed didn't move
- machine off
- moved bed to y min
- machine on
- 6th print - successful
- 7th print - G26 mesh validation ran
- machine off
- machine on
- 8th print - successful
- 9th print - successful
- 10th print - G26 mesh validation ran
- machine off

marciot, 2020-09-14:
I made a new testing release with G26 both enabled and disabled, for comparison. For the one with G26 enabled, I added a check to skip the command if it appeared to be running in start GCODE (i.e. after M75), to prevent the issue with Lulzbot start gcode.

The release is here:

https://github.com/drunken-octopus/drunken-octopus-marlin/releases/tag/v2.0.0.1rc23

IggyTheWolf, 2020-09-15:
Awesome! I could replicate the runaway or even a crash. 

The G26 is a neat feature. It keeps me from running my own layer test. I like that you're keeping it.

marciot, 2020-09-15:
You could replicate the runaway on which one?

marciot, 2020-09-15:
@IggyTheWolf: Since the runaway issue is a separate issue from this one, could I ask you to create a new git hub issue with the details? Please include:

- Exactly under which firmware builds you were able to reproduce the problem
- Under which firmware builds you were not able to reproduce the problem
- The exact sequence of steps after power on you took to reproduce the problem
- Whether the problem happens in a repeatable fashion every time, or just occasionally.

Once I have that info I can see if I can reproduce it and begin to identify the differences between the builds with the problem and the builds without.

IggyTheWolf, 2020-09-15:
Sorry could not replicate. I couldn't even reproduce a crash like before.

marciot, 2020-09-15:
> Sorry could not replicate. I couldn't even reproduce a crash like before.

@IggyTheWolf: That's good news... I suppose? Well, we'll keep an eye on it. Hopefully it was just a fluke.

IggyTheWolf, 2020-09-15:
I will. I did a 3hr print with the modified start code (I commented out more things) and it worked just fine. 

I haven't been able to get a crash on the Marlin boot screen. If you ever see that I think you're close.

SwiftNinja, 2020-10-08:
I'm not super sure what the end result of this ticket is, but I'm sort of along the same lines as @mariot when he suggested modifying your startup gcode instead of modifying the firmware.

G26 is clearly documented in marlin as "Mesh Validation Pattern". If Lulzbot was misusing G26 for other purposes because it had other side effects, than, for me, that's a mistake on Lulzbot's side, not Marlin's side.

If we're talking about scrapping G26's implementation and simply returning it to a 'flush error buffer', that would seem to me to negatively impact people who want to use UBL.

https://marlinfw.org/docs/gcode/G026.html

Am I missing something?

marciot, 2021-12-14:
@IggyTheWolf: It looks like I had a workaround for this in release 22, but it got lost in recent versions. I have reinstated the fix and it will be available on the next release.