drunken-octopus/drunken-octopus-marlin #36: Archim 2 SD card

shimmytoast, 2022-01-24:
I'd like to use the onboard SD card slot on the Archim 2. I'm currently struggling to get that working. 

I'm using the Taz Pro configuration with the LCD controller changed to the FT810 for a custom TAZ 5 "Pro" build.

shimmytoast, 2022-01-24:
[image]

marciot, 2023-04-22:
@shimmytoast: I suspect the issue is that the graphical display uses the hardware SPI bus, which is also used for the onboard SD card slot, hence, the two things are mutually exclusive.

Marlin does not support using software SPI for the SD card, so the only alternative would be to run the display off the other display connector and use software SPI for it. This configuration is less tested, but has been used in some prototype printers. You need to change AO_EXP2_PINMAP to AO_EXP1_PINMAP in the configs and then wire it according to this:

https://github.com/drunken-octopus/drunken-octopus-marlin/blob/74d65ecdf3f01b34f1ddf95f3d1bde9f5d79f3bd/Marlin/src/lcd/extui/ftdi_eve_touch_ui/pin_mappings.h#L74

This will liberate the hardware SPI for use by the SD card code.