drunken-octopus/drunken-octopus-marlin #42: [BUG][Workhorse | M175]

HiTekCowboy, 2022-12-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

X and Y axis are skewed.
Home Position shows as -57(-56.9), -17(16.9), 299
Should be -50, -17, 299

Causes issues with Auto-Calibration, Nozzle-Wiping, Bed-Leveling.

### Bug Timeline

R53

### Expected behavior

Normal Auto-Calib, Nozzle Cleaning, Probing/Leveling.

### Actual behavior

1. On running Auto-Calibrate(or anything that calls for nozzle_wipe) the nozzle stops to heat(soften) about +6-7mm on the X(to the right) of where it should be to lower and start wiping. 
2. When bed leveling, the nozzle similarly lands about 6-7mm off from where it should land on the front-left leveling washer.
3. When attempting an M48 Probe test(after adjusting my Z offset) I'm getting a "Probe out of bounds". 

### Version of Marlin Firmware

bugfix 2.1.X

### Printer model

Lulzbot Workhorse M175v2

### Electronics

Archim2 board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

My theory is that in 'build-config.py' **lines 1165 - 1168** are the culprit for these issues:

        TOOLHEAD_X_MIN_ADJ                               = -6.9
        TOOLHEAD_X_MAX_ADJ                               = -6.5
        TOOLHEAD_Y_MIN_ADJ                               =  0.1
        TOOLHEAD_Y_MAX_ADJ                               = -13.5

I'm assuming these X and Y adjustments may be necessary for the M175 in some other printers. 
For Workhorse builds, these adjustments should be omitted.
Comparatively tested my HE(Goldenrod) toolhead with it's firmware(which functions as normal) to rule out other possibilities.

marciot, 2022-12-02:
@HiTekCowboy: So I did some digging on this and found out that those offsets were added on the suggestion of mythikwolf in the following thread:

https://forums.drunkenoctop.us/t/which-firmware-for-m175-toolhead/59

In that thread, he says the nozzle of that tool head sits at a slightly different position than other tool heads and thus the offsets are necessary for the tool head. In this thread, you are proposing those offsets are machine specific, rather than tool head specific. It can't be both ways, so something does not quite add up.

I had asked mythikwolf to read off the offsets from LulzBot firmware, which he did so, so unless their firmware has changed, the values should be correct.

You may want to verify the offsets from official LulzBot firmware yourself, to see if you are getting the same results as mythikwolf. Load up the LulzBot specific firmware and then give me the axis limits, as described in this post:

https://forums.drunkenoctop.us/t/which-firmware-for-m175-toolhead/59/10?u=marciot

With that information, we might at least be able to get a better understanding of what is happening.

marciot, 2022-12-02:
I did find this. I wonder if the offsets are different for the v1 vs the v2 toolheads. Are you maybe able to confirm which version you have?

https://lulzbot.com/store/m175-v2-conversion-kit-retail-packaged-kt-pp0055?ref=KT-PP0055

HiTekCowboy, 2022-12-02:
@marciot My toolhead is the M175 v2.

I probably should have mentioned that in testing, when parking to heat E1 for wipe, over the wipe pad, the nozzle is roughly 6-7mm off from where it is supposed to be (photo), which is evidence of the issue.
[image]
Further, after looking at the two toolhead assemblies in solidworks, and doing some quick hard measurements, I ran the M175(disconnected from the 20pin to be safe) through a wipe to the point where it would briefly park(before alarming and rehoming due to not getting a signal from the T1) on the Goldenrod firmware, and the nozzle lined up exactly where it was supposed to be normally otherwise.
I suspect that mythikwolf was running a M175v1, as v1 was designed before the Universal Tool Head came into play, and that likely required having the offset in question.

HiTekCowboy, 2022-12-03:
Went back and reinstalled the Stock 1.4 board with stock firmware, and followed the instructions you gave to mythikwolf as well. The following are my values.

X-min: -50
Y-min: -17
X-max: 293
Y-max: 308

XYZ after homing:
X = -48, Y = -15, Z = 297

marciot, 2022-12-05:
@HiTekCowboy: That checks out. So it looks like I just need to create two toolhead variants, a M175v1 with the extra offsets, and a M175v2 without. I'll add this to the next release. Thank you for your work on this! I appreciate it!

HiTekCowboy, 2022-12-05:
@marciot Not a problem!
I've already done a quick and dirty build and tested this proposed change, as well as a few other changes that apply to the Workhorse as a whole, and I seem to have solved several issues. 
I additionally just tweaked the Workhorses Min/Max, as well as the Calibration Cube parameters, as the original values seemed to be those of the Pro, and were causing the bed to try and move past it's real max during calibration.
I'll do a proper write up later this evening or tomorrow, and set up a proper git clone to show my changes, and make life easier for you. 
I apologize for my dinky way of going about and explaining things, I mainly only code as a hobby these days, and know enough to be dangerous, but I haven't messed with GitHub much, otherwise I would have done all this from the get-go to try and be more helpful.

marciot, 2023-04-22:
@HiTekCowboy: Can this ticket be closed? I believe a fix was put into release 55

HiTekCowboy, 2023-04-22:
> @HiTekCowboy: Can this ticket be closed? I believe a fix was put into release 55

Done, sorry about that.
Thought I closed it.

closed 2023-04-22 by HiTekCowboy