drunken-octopus/drunken-octopus-marlin #43: [BUG] TAZ6_DualExtruderV3 M301 ignores E0,E1 Parameter

b-morgan, 2022-12-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The Marlin_Oliveoil_TAZ6_Yellowfin_DualExtruderV3_bugfix-2.1.x.1_21539745c4.hex firmware M301 ignores the E0 or E1 parameter and sets both extruders to the same value.
```
Send: M301 E0 P37.02 I2.95 D116.01
Recv: ok P15 B4
Send: M301
Recv:   M301 P37.02 I2.95 D116.01
Recv:   M301 P37.02 I2.95 D116.01
Recv: ok P15 B4
Send: M301 E1 P34.62 I2.41 D124.06
Recv: ok P15 B4
Send: M301
Recv:   M301 P34.62 I2.41 D124.06
Recv:   M301 P34.62 I2.41 D124.06
Recv: ok P15 B4
```

### Bug Timeline

I don't know, this is the first time I've used the Dual Extruder V3 with this firmware

### Expected behavior

I expected to be able to set the PID values for each extruder separately.

### Actual behavior

It sets both PID values to the last one entered.

### Steps to Reproduce

1. Send M301 E0 P37.02 I2.95 D116.01
2. Send M301 E1 P34.62 I2.41 D124.06
3. Send M301

### Version of Marlin Firmware

Send: M115 Recv: FIRMWARE_NAME:Marlin bugfix-2.1.x (Dec 17 2022 14:06:36) SOURCE_CODE_URL:https://github.com/marciot/drunken-octopus-marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:TAZ 6 EXTRUDER_COUNT:2 UUID:845f003c-aebd-4e53-a6b9-7d0984fde609 Recv: Cap:SERIAL_XON_XOFF:0 Recv: Cap:BINARY_FILE_TRANSFER:0 Recv: Cap:EEPROM:1 Recv: Cap:VOLUMETRIC:0 Recv: Cap:AUTOREPORT_POS:0 Recv: Cap:AUTOREPORT_TEMP:1 Recv: Cap:PROGRESS:0 Recv: Cap:PRINT_JOB:1 Recv: Cap:AUTOLEVEL:1 Recv: Cap:RUNOUT:1 Recv: Cap:Z_PROBE:1 Recv: Cap:LEVELING_DATA:1 Recv: Cap:BUILD_PERCENT:1 Recv: Cap:SOFTWARE_POWER:0 Recv: Cap:TOGGLE_LIGHTS:0 Recv: Cap:CASE_LIGHT_BRIGHTNESS:0 Recv: Cap:EMERGENCY_PARSER:1 Recv: Cap:HOST_ACTION_COMMANDS:1 Recv: Cap:PROMPT_SUPPORT:1 Recv: Cap:SDCARD:1 Recv: Cap:MULTI_VOLUME:0 Recv: Cap:REPEAT:0 Recv: Cap:SD_WRITE:1 Recv: Cap:AUTOREPORT_SD_STATUS:0 Recv: Cap:LONG_FILENAME:0 Recv: Cap:LFN_WRITE:0 Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0 Recv: Cap:EXTENDED_M20:0 Recv: Cap:THERMAL_PROTECTION:1 Recv: Cap:MOTION_MODES:0 Recv: Cap:ARCS:1 Recv: Cap:BABYSTEPPING:1 Recv: Cap:CHAMBER_TEMPERATURE:0 Recv: Cap:COOLER_TEMPERATURE:0 Recv: Cap:MEATPACK:0 Recv: Cap:CONFIG_EXPORT:0 Recv: ok P15 B4 Send: M155 S5 Recv: ok P15 B4 Send: M876 P1 Recv: ok P15 B4

### Printer model

LulzBot TAZ6

### Electronics

Stock

### Add-ons

None

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

b-morgan, 2022-12-22:
I didn't compile the firmware, @marciot did, so there's no .zip of the configuration files, I just checked the box so I could enter the issue.

marciot, 2022-12-22:
@b-morgan This is because both sides of the dual are physically the same heater block, the same heater, the same thermal mass, etc. and hence should not have extruder-specific PID values,

marciot, 2022-12-22:
@b-morgan It is possible to configure Marlin to have separate values using PID_PARAMS_PER_HOTEND, but this is only necessary when the two extruders are mismatched.

b-morgan, 2022-12-22:
I don't know about your LulzBot Dual Extruder V3 but mine has two heater blocks and two thermistors. I PID tuned both (M303 takes an E parameter) and got the different values shown in my initial report (the default value is P47.45 I4.83 D116.63 for both). The V3 does have only one cooling fan because the cold side is shared.

I believe you should set PID_PARAMS_PER_HOTEND in your config.

(BTW, The LulzBot Dual Extruder V2 also has two heater blocks and two thermistors about 50mm apart).

marciot, 2022-12-22:
@b-morgan You are correct, each side has a heater block and thermistor, but they are the same heater blocks and thermistors on both sides, hence, you only need a single PID value. You would set PID_PARAMS_PER_HOTEND if you had a non-matched heater and thermistor, for example, if you had a toolhead with an E3D Hermera on one side, and a Mosquito on the other.

It's not surprising you got different PID values on both sides, however; if you do this multiple times on the same tool head, you will get values that differ each time. It's not an exact value.

b-morgan, 2022-12-22:
Then I'm confused by what Marlin reports. Using your argument would imply that M301 with no parameters and PID_PARAMS_PER_HOTEND not defined should output a single value, not a value for each extruder.

The firmware supplied by LulzBot allows separate PID values to be set so I believe you should allow it also. Its not like you have to add any code, just set the define that is already available. 

It would take me hours to figure out how to set it and compile my own copy 😢. 
I'll bet you would spend less than 5 minutes😀.

marciot, 2022-12-22:
@b.morgan: What evidence do you have that the PID needs to be tuned differently from the default values? The PID need only be adjusted when the temperature is either overshooting the desired value, or is taking a long time to reach it. See the following graph:

[image]

Adjusting the PID is a coarse adjustment for correcting undershooting or overshooting. Unless you have verified this particular issue, the PID values can be left at their defaults.

marciot, 2022-12-22:
@b-morgan: I don't object to adding the capability you desire in the next release, but I also suspect it is unnecessary.

b-morgan, 2022-12-22:
I purchased an M175 tool head for my TAZ 6 (which I then upgraded it to an M175V2). The default PID values in the firmware supplied for that tool head was so far off that it wouldn't stabilize and the start gcode hung on the first set temperature command (A later release of the firmware corrected the issue). The fix, of course, was to PID tune that tool head and store the better values in the EEPROM.

Curiosity got the best of me so I PID tuned all of my tool heads and added M301 commands to the start gcode of those with values that were "different" than the defaults. My definition of "different" was values that varied from the defaults by about the same amount that the failing and working PID values differed on the M175V2. As you can see from the data in this issue, I decided that "P37..." and "P34..." were "different" than the default of "P47..." for the Dual Extruder V3. I will admit that this isn't a very scientific approach.

You are, of course, arguing that "P37..." and "P34..." are close enough that either set of parameters should work (and you are probably correct).

Since I've finished printing my granddaughter's Christmas present, I can run a few PID tunes and see just how much variance there actually is.

Adding that define in the next release is perfectly acceptable to me. I believe it will be less confusing for me and possibly others as well.

Thanks for taking the time to discuss this with me.

closed 2023-04-17 by b-morgan